WMXZ-EU/MTP_t4 #8: Teensy 3.2

Fluxanode, 2020-11-29:
I have tried really hard and cannot get this to work on a teensy 3.2.  I have the latest version of Teensyduino, done all of the the mods as per instructions.  It compiles the source files with no issues but is does no work like the original code.  It is not showing up in Win explorer.  Is there some place to set the CS pin? I am using the SPI and change made the changes for that.  Can you help me out with some ideas?

WMXZ-EU, 2020-11-29:
If you use mtp_test example, you have to modify the SPI settings SD_MOSI etc, sd_str[] and cs[] to reflect your configuration.
All this is in function store_configure(), which is called from setup

Fluxanode, 2020-11-30:
I don't see store_configure() in setup.  Maybe usb_mtp_configure()?  What file is the function in so i can edit?
My setup function in mpt_test.

void setup()
{ 
  while(!Serial && millis()<3000); 
  usb_mtp_configure();
  if(!Storage_init()) {Serial.println("No storage"); while(1);};

  Serial.println("MTP test");

#if USE_SDIO==1
  pinMode(13,OUTPUT);
#endif

Fluxanode, 2020-11-30:
Ok I just updated to the latest.  So for spi use_sd is 1?

WMXZ-EU, 2020-12-01:
T3.2 has no BUILTIN_SDCARD , which is a 'pseudo CS for T3.6
updated example should compile also on T3.2

Fluxanode, 2020-12-02:
I tried to compile the new code and now am seeing these errors...

D:\ChuckW\Arduino\libraries\MTP_t4-master\src\MTP.cpp:744:8: error: prototype for 'void MTPD::SendObject()' does not match any in class 'MTPD'
   void MTPD::SendObject() {
          ^
In file included from D:\ChuckW\Arduino\libraries\MTP_t4-master\src\MTP.cpp:28:0:
D:\ChuckW\Arduino\libraries\MTP_t4-master\src\MTP.h:127:8: error: candidate is: bool MTPD::SendObject()
   bool SendObject() ;
           ^
D:\ChuckW\Arduino\libraries\MTP_t4-master\src\MTP.cpp: In member function 'void MTPD::loop()':
D:\ChuckW\Arduino\libraries\MTP_t4-master\src\MTP.cpp:798:27: error: no matching function for call to 'MTPD::openSession()'
               openSession();
                                    ^
D:\ChuckW\Arduino\libraries\MTP_t4-master\src\MTP.cpp:601:10: note: candidate: void MTPD::openSession(uint32_t)
     void MTPD::openSession(uint32_t id)
             ^
D:\ChuckW\Arduino\libraries\MTP_t4-master\src\MTP.cpp:601:10: note:   candidate expects 1 argument, 0 provided

Error compiling for board Teensy 3.2 / 3.1.

Fluxanode, 2020-12-02:
Question - How did you determine the value for #define SPI_SPEED SD_SCK_MHZ(33)  // adjust to sd card ?

WMXZ-EU, 2020-12-03:
The spi speed depends on your hardware 12 MHz should always work, but there is no rule. simply try it until it works.
To change the name of the portable device you have to modify the MTP code.
On drive letter, windows does not assign a drive letter to portable devices. 
If you wanted to change the disk name in the file explores, there is a sd_str string in the example that one can change

closed 2022-03-23 by WMXZ-EU