WMXZ-EU/MTP_t4 #11: Compiling mtp-test.ino for Teensy 3.6

hystrix1, 2021-04-22:
I am trying to compile mtp-test-ino for Teensy 3.6 using Arduino IDE + Teensyduino.

I get the error: **'FS' has not been declared** in storage.cpp at this line:

`void sd_addFilesystem(FS &fs, const char *name) {`

I am using the following libraries:
Using library SD at version 1.2.2 in folder: /home/user/arduino-1.8.13/hardware/teensy/avr/libraries/SD 
Using library Time at version 1.6 in folder: /home/user/arduino-1.8.13/hardware/teensy/avr/libraries/Time 
Using library SPI at version 1.0 in folder: /home/user/arduino-1.8.13/hardware/teensy/avr/libraries/SPI 

If I change `#include "SD.h"` to `#include "SdFat.h"` and use the SDFat V2 library from here [https://github.com/greiman/SdFat](url), I get the same error.

What am I missing?

WMXZ-EU, 2021-04-22:
You need the latest TD that is modified for SdFat V2. and has SD Version 2.0.0

hystrix1, 2021-04-22:
I have TD version 1.53 for Linux X86 64 bit. This appears to be the latest version.

WMXZ-EU, 2021-04-22:
you need TD1.54b7

closed 2021-04-22 by WMXZ-EU