WMXZ-EU/MTP_t4 #14: MTP device not in windows explorer, serial port not available

Fluxanode, 2021-06-10:
I installed MTP_T4 on the Teensy 3.5 and made the modifications to the usb_desc.h. 
I am using arduino IDE with Teensyduino version 1.54-beta 9
1. The device shows up in the device manager as MTP USB Device not Teensy MTP device.
2. It is not showing up in Windows explorer.
3. After making the changes to USB_DESC should there be a Serial device in device manager for the Teensy 3.5?

I am going to also post this to the pjrc forum but though you may like a note here.

Fluxanode, 2021-06-21:
FYI No help on PJRC forum.  MPT APPARENTLY 
does not work on 3.5 teensy.  
Thanks for the info...

WMXZ-EU, 2021-06-24:
MTP does work on T3.5. Just compiled and run it.
compiled with USB type "MTP Disk Serial (Experimental)".

In Device Manager portable device shows as Teensy
USB Serial Com port is accessible.

Note: if you declare multiple sd devices (as with mtp-test.ino) : sdx[0] MUST exits. 
If you are using jumper cables to connect micro sd boards, the spi speed (33 MHz) may be too high. reduce to say 15 MHz.

Fluxanode, 2021-06-24:
Thanks for the reply.  
I am using a new T3.5 with arduino 1.8.15 and Teensyduino-beta9 with the built-in SD.  Both with a fresh install.  I'm  not sure what you mean "if you declare multiple sd devices (as with mtp-test.ino) : sdx[0] MUST exits." I'm only using the one onboard SD with the out of the box install, do I need to change something in the test program?
What versions did you test with and what example program are you using?  I have followed the install directions to the letter and tested using the included MPT-test.ino. 

Can you review the discussion on the PJRC forum MTP responder? You can see I worked with Defragster on this and verified the SDfat is working and my install is good and working, at least the SD part...

Fluxanode, 2021-06-24:
Is there a specific USB driver to use?  I set up on another computer (without any Arduino or Teensyduino software) and at first the teensy was there as Teensy under an unknown device until the device drivers finished loading/installing then it moved to portable devices as MTP USB DEVICE with a yellow ! next to it and the error message "Device driver software was not successfully installed" came up.  BTW the serial port only displays in Device Manager when the port is set to serial.  Goes away when set to MTP disk and programed. 

Maybe a driver issue?  
Or teensyduino? I just don't know.