WMXZ-EU/MTP_t4 #21: Teensy 4.1 not showing as MTP storage device

asinghnu, 2022-04-06:
Teensy 4.1  is used as data logger ,however after doing the changes in files as mentioned in MTP_t4 there is no Teensy as storage device
although the data is updated in sdcard (32 GB). the same changes done in other computer shows it as storage but when i upload program from my computer it fails as storage .system specifications are Teensy 4.1,windows 10,Teensyduino, Version 1.5,arduino-1.8.19

WMXZ-EU, 2022-04-06:
Does MTP device show up with mtp_basic?
using MTP during and after logging is very tricky, as MTP is designed mainly for file transfers that exist when MTP is initiated.

asinghnu, 2022-04-06:
my port on Arduino shows as "COM 20 Serial Teensy ",not sure whether it shows up with MTP_basic,device manager detects as USB .can you please guide with all modifications required

WMXZ-EU, 2022-04-06:
You must set USB type to MTP disk (either serial or not)
Port will shoe up as Everything

asinghnu, 2022-04-06:
port is showing as "hid#vid_16c0&pid_04d1 MTPDisk(Teensy4.1) when selected usb as mtp disk.still the no storage device present

WMXZ-EU, 2022-04-06:
Just compiled mtp-basic for T3.6 and T4.1 and no issue
(added "include <SPI.h>" at top of file, see actual github file)

closed 2022-04-13 by asinghnu