WMXZ-EU/MTP_t4 #22: Files read through GetPartialObject return some garbage

james-portman, 2022-11-09:
The easiest way to replicate is using a very small text file e.g. "hello"

Some applications will only read the correct amount of data according to the file size, but some load in everything that the code sends, which might be a minimum of MTP_TX_SIZE packet

I think this might also happen with large files which are not exactly a multiple of MTP_TX_SIZE

WMXZ-EU, 2022-11-10:
Thanks for finding that and PR

james-portman, 2022-11-10:
No problem.

I was trying to work out why this was running so badly with Linux but I think it's just a Linux issue.
It is about 50/50 chance whether the drive mounts, then seems like the maximum file size I can load is about 100KB, any more and it starts to lock up.

It seems pretty good with Windows 10 when I tried, and quite quick too.

I'll do some more debugging on the Linux side and raise new issues/PRs if I find anything that I can describe more specifically.