WMXZ-EU/MTP_t4 #24: Different filesize when copy from MTP device

Anjali8340, 2025-02-07:
I am using SdFat library to write data into sd card, then i put teensy into MTP mode. when i copy a file from sd card to my computer the filesize increment. Like original file is 16kb but copied file is 17kb. The bytes at the end of the file is copied once again inside the file

WMXZ-EU, 2025-02-07:
Just copied a file from Teensy4.1 to PC (Win11) via MTP (using mtp-basic.ino, from MTP_t4/examples)
On uSD card: Size: 2 458 112 bytes ; Size on Disk: 2 621 440 bytes
on PC:            Size: 2 458 112 bytes ; Size on Disk: 2 461 696 bytes

File site is identical on uSD card and transferred via MTP, That the 'on Disk' sizes are different is due to different cluster sizes and file systems, but what counts is the true file size.
I checked the sizes using file properties of file on SD card and on MTP-transferred file
 
So cannot reproduce the issue. 
Can you try again with only mtp-basic.ino and check with file properties?

Anjali8340, 2025-02-10:
i logging data on sd card. I create .hex file of 16MBytes. When i am copy single file the file size is same, but when i try to copy multiple files each copied file has extra 512 bytes at the end and these 512 bytes are just copy of last 512 bytes of actual file. 
i have checked file size with file properties and this is not about file size, the copied file actually has 512 extra bytes.

One more thing i log data for hours and then i put teensy into mtp mode (by calling mtpd.loop() in loop). But the sd card does not appear, i have to restart my board and put it into mtp mode. i am calling addfilesystem() function in setup as you did. can you suggest me some possible reason

WMXZ-EU, 2025-02-10:
This first point needs more investigation.
To the second point.:
After logging data to SD card, the PC must be instructed to refetch the data. On Linux one would unmount and remount the mtp folder. on Windows, one could unmount and remount also. there is a PS script that should do the. What I usually do is to send the command 'r' to program that sends a reset event to PC (see mtp-logger.ino line 349). Another way is instead of sending an event to PC is to toggle the USB connection
For example

```
  extern "C" void usb_init(void);
  void resetUSB(void)
  {
    USB1_USBCMD = 0;  // disconnect USB
    delay(100);       // long enough for PC+hubs to detect
    usb_init();
    #if USE_MTP==1
      while(!Serial) mtpd.loop();
    #endif
  }
```

WMXZ-EU, 2025-02-11:
Corrected the getObject method.  Own test do transfer only the correct number of bytes and do not repeat the last 512 bytes (USB buffer for T4)

Anjali8340, 2025-02-12:
I will test with updated code.
and for MTP device not appearing Resetting USB worked. Thanks

Anjali8340, 2025-02-13:
So with this new code the problem of different filesize is resolved, but a new problem has occured. As i told you I am generating files of 16MB, number of files depends on how long data logging went. Now when i try to copy multiple files at once, like 50 files of 16MB, it copies 3 to 4 files then it shows error "The requested value cannot be determined". I am sharing you a screenshot of this error. After this i am able to copy even a single file. 

[image]

WMXZ-EU, 2025-02-13:
Did another correction and was able to copy a folder with 60 files in it, each 10 MB in size.

Anjali8340, 2025-02-14:
Problem solved. Thanks

closed 2025-02-17 by Anjali8340